lis-epfl/robogen #56: Attitude of Mass of Cylinder and Capsule Wrong

basil-huber, 2016-10-20:
In the functions Model::addCylinder(...) and Model::addCapsule take the direction of the axis of the cylinder. We pass this direction to dMassSetCylinderTotal. We then set the cylinder's attitude depending on the orientation. This means we first set the orientation of the mass correct and then turned it to be wrong.

    boost::shared_ptr<SimpleBody> Model::addCylinder(float mass,
    		const osg::Vec3& pos, int direction, float radius, float height,
    		int label) {
    	dMass massOde;
    	dMassSetCylinderTotal(&massOde, mass, direction, radius, height);
    	dxGeom* g = dCreateCylinder(this->getCollisionSpace(), radius, height);
    	osg::Quat rotateCylinder;
    	if (direction == 1) {
    		rotateCylinder.makeRotate(osg::inDegrees(90.0), osg::Vec3(0, 1, 0));
    	} else if (direction == 2) {
    		rotateCylinder.makeRotate(osg::inDegrees(90.0), osg::Vec3(1, 0, 0));
    	}

    	boost::shared_ptr<SimpleBody> body(new SimpleBody(shared_from_this(),
    				massOde, g, pos, rotateCylinder));
    	this->addBody(body, label);
    	return body;

jauerb, 2016-10-20:
> repository","main_image_url":"https://cloud.githubusercontent.com/assets/143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png","action":{"name":"Open
> GitHub","url":"https://github.com/lis-epfl/robogen"}},"updates":{"snippets":[{"icon":"DESCRIPTION","message":"Attitude
> of Mass of Cylinder and Capsule Wrong (#56)"}],"action":{"name":"View
> Issue","url":"https://github.com/lis-epfl/robogen/issues/56"}}}

Perhaps you are right?  I would have to take a closer look.  Do you have 
a demo showing the wrong behavior?

basil-huber, 2016-10-20:
I'll upload one tomorrow

basil-huber, 2016-10-21:
I uploaded a demo of the bug on the following branch:
https://github.com/basil-huber/robogen/tree/test/inertia

To execute, just run ./inertia-test;   In the test case, a torque is applied to a cylinder along different axis. The cylinder is expected to rotate fastest around its axis of symmetry (z) and slow around the other axis. However, it rotates fastest around the x axis, indicating that the moment of inertia has its axis of symmetry along the x axis.